ISEE-Technology/CamVox #6: error when ./build.sh

h-k8888, 2020-12-12:
because of the wrong VTK version?

/usr/include/vtk-6.2/vtkObjectBase.h:165:30: **error:** field ‘ReferenceCount’ has incomplete type ‘vtkAtomicInt<int>’
   vtkAtomicInt<vtkTypeInt32> ReferenceCount;
                              ^
In file included from /usr/include/vtk-6.2/vtkObjectBase.h:44:0,
                 from /usr/include/vtk-6.2/vtkSmartPointerBase.h:27,
                 from /usr/include/vtk-6.2/vtkSmartPointer.h:23,
                 from /usr/include/pcl-1.7/pcl/visualization/point_cloud_geometry_handlers.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:40,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc:7:
/usr/include/vtk-6.2/vtkAtomicInt.h:307:7: note: declaration of ‘class vtkAtomicInt<int>’
 class vtkAtomicInt: public detail::vtkAtomicIntImpl<T>
       ^
In file included from /home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc:7:0:
/usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^
In file included from /usr/include/c++/5/memory:81:0,
                 from /usr/include/c++/5/thread:40,
                 from /home/hk/ws_CamVox/src/CamVox/isee-camvox/include/System.h:19,
                 from /home/hk/ws_CamVox/src/CamVox/isee-camvox/include/pointcloudmapping.h:4,
                 from /home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc:1:
/usr/include/c++/5/bits/unique_ptr.h:49:28: note: declared here
   template<typename> class auto_ptr;
                            ^
/home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc: In member function ‘void Camvox::PointCloudMapping::updatecloud()’:
/home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc:168:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int i = 0; i < currentvpKFs.size(); i++)
                               ^
/home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc:170:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 for (int j = 0; j < pointcloud.size(); j++)
                                   ^
/home/hk/ws_CamVox/src/CamVox/isee-camvox/src/pointcloudmapping.cc:172:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                     if (pointcloud[j].pcID == currentvpKFs[i]->mnFrameId)
                                            ^
CMakeFiles/Camvox.dir/build.make:315: recipe for target 'CMakeFiles/Camvox.dir/src/pointcloudmapping.cc.o' failed
make[2]: *** [CMakeFiles/Camvox.dir/src/pointcloudmapping.cc.o] Error 1
CMakeFiles/Makefile2:98: recipe for target 'CMakeFiles/Camvox.dir/all' failed
make[1]: *** [CMakeFiles/Camvox.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2
Converting vocabulary to binary
./tools/bin_vocabulary: error while loading shared libraries: libDBoW2.so: cannot open shared object file: No such file or directory

xuankuzcr, 2020-12-14:
Trying to uncomment this line `#include <pcl/visualization/pcl_visualizer.h>` in `Calibrating.h` might solve the problem.

h-k8888, 2020-12-15:
> Trying to uncomment this line `#include <pcl/visualization/pcl_visualizer.h>` in `Calibrating.h` might solve the problem.

The problem is solved, thanks.

closed 2020-12-15 by h-k8888