ISEE-Technology/CamVox #9: Can this work be used with other lidars and cameras?

xiaobrnbrn, 2020-12-28:
Hello, thanks for your excellent work. I would like to test this work on my own sensors, such as the RoboSense lidar and ZED camera, is this feasible? If possible, how do I modify it?

zywok, 2021-01-08:
Thank you for your interest in our work. Benefit from the non-repetitive scanning method of Livox lidar, which makes the point cloud accumulate in the FOV over time. The relatively dense depth information of a single frame and the accumulated dense depth information of multiple frames make mapping and calibration possible. The gap between multiple lines of RoboSense lidar cannot be filled over time, so the edge matching method cannot be used in calibration. However, you can try to apply it to tracking and mapping.

biao-y, 2021-03-29:
> 您好，感谢您的出色工作。我想在自己的传感器（例如RoboSense激光雷达和ZED相机）上测试这项工作，这可行吗？如果可能的话，我该如何修改？

Hello, my lidar, like the robosense you said, I also want to use it to fuse with the camera. Have you solved this problem now? I'm not sure as a beginner. Thank you very much for your help!

closed 2021-04-27 by zywok