ISEE-Technology/CamVox #12: The dataset in SUSTech campus with loop closure does not contain lidar data

daijicheng, 2021-01-08:
The bag only contains rgb image and depth image, but I hope to learn how to undistorted point cloud using imu data and project the lidar point cloud to image. 
will this part of code be released later？

zywok, 2021-01-08:
Thank you for pointing out the problem. The code for removing the distortion of the point cloud and generating the depth map is under the isee-preprocessing folder. The Rosbag files for testing online calibration contains point cloud information. Because I was disturbed by the application for Phd some time ago, and next I will optimize the overall implementation of the code and will continue to maintain it. The loop map dataset with point clouds will be updated as soon as possible. Thank you for understanding.

daijicheng, 2021-01-08:
Thank you so much for your quick reply. When using the calibration dataset，I encountered a problem that the saved point cloud, calibration.pcd, is not transformed into consistent coordinate.
[image]
It seems caused by that the function LivoxMsgCbk is not completed.  I am very looking forward to the optimized code.

closed 2021-04-27 by zywok