ISEE-Technology/CamVox #14: The problem Ubuntu18.04 compiling

waynehq, 2021-01-12:
I have encountered problems compiled in Ubuntu18.04,it seems to need PCL 1.7 for the project,but PCL 1.8 in ros melodic,how can i do？Thank you for your help!

JacoBasil, 2021-03-22:
Hi, guys!
Maybe I have met the same issue when I tried to compile ROS nodes file ./build_ros.sh in /isee-camvox folder. I have got PCL-1.8 version, but it was permanently working with other Livox repositories needed PCL-1.7. Anyway I installed PCL-1.7 and boost-1.58, but the problem haven't been solved. 

This is an error message:
[100%] Linking CXX executable ../camvox
/usr/bin/ld: warning: libboost_system.so.1.65.1, needed by /opt/ros/melodic/lib/libroscpp.so, may conflict with libboost_system.so.1.58.0
/usr/bin/ld: warning: libboost_filesystem.so.1.65.1, needed by /opt/ros/melodic/lib/libroscpp.so, may conflict with libboost_filesystem.so.1.58.0
/usr/bin/ld: warning: libboost_thread.so.1.58.0, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_common.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_kdtree.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_search.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_filters.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_io.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_visualization.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_segmentation.so.1.7, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkRenderingFreeTypeFontConfig-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkRenderingMatplotlib-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkRenderingFreeTypeOpenGL-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libGLEW.so.1.13, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkInteractionStyle-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkRenderingFreeType-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkRenderingOpenGL-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkRenderingCore-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkCommonDataModel-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkCommonMath-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libvtkCommonCore-6.2.so.6.2, needed by ../../../lib/libCamvox.so, not found (try using -rpath or -rpath-link)
../../../lib/libCamvox.so: undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
../../../lib/libCamvox.so: undefined reference to `boost::this_thread::hiden::sleep_until(timespec const&)'
../../../lib/libCamvox.so: undefined reference to `vtkRenderingFreeTypeOpenGL_AutoInit_Destruct()'
../../../lib/libCamvox.so: undefined reference to `vtkRenderingFreeTypeOpenGL_AutoInit_Construct()'
collect2: error: ld returned 1 exit status
CMakeFiles/camvox.dir/build.make:462: recipe for target '../camvox' failed
make[2]: *** [../camvox] Error 1
CMakeFiles/Makefile2:343: recipe for target 'CMakeFiles/camvox.dir/all' failed
make[1]: *** [CMakeFiles/camvox.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

zywok, 2021-04-27:
[image]
Feeling so sorry for the slow rely. I think Compilation errors occur, most of which are caused by PCL library problems ,we have update the PCL library install method and supply the installation package link, you can try it.

closed 2021-04-27 by zywok