ISEE-Technology/CamVox #16: fatal error: pcl/common/transforms.h: 没有那个文件或目录

alsosos, 2021-01-15:
./build_ros.sh时报错，请问怎么解决

dongfangzhou1108, 2021-03-11:
我也是，hxd，请问你弄明白了没，谢谢

alsosos, 2021-03-11:
你ubuntu是什么版本

alsosos, 2021-03-11:
pcl是什么版本 好像是pcl版本问题

dongfangzhou1108, 2021-03-13:
确实，我的是ubuntu18，作者的是16吗，要pcl1.7

alsosos, 2021-03-13:
linux版本没问题 pcl需要1.7的 之前我高版本也不行

dongfangzhou1108, 2021-03-14:
多谢，我再试试

dongfangzhou1108, 2021-03-14:
兄弟，我安装了pcl1.7,但仍然有错误：
../../../lib/libCamvox.so：对‘pcl::search::Search<pcl::PointXYZ>::getName() const’未定义的引用
../../../lib/libCamvox.so：对‘boost::this_thread::hiden::sleep_until(timespec const&)’未定义的引用
../../../lib/libCamvox.so：对‘vtkRenderingFreeTypeOpenGL_AutoInit_Destruct()’未定义的引用
../../../lib/libCamvox.so：对‘vtkRenderingFreeTypeOpenGL_AutoInit_Construct()’未定义的引用
你怎么解决的？谢谢

alsosos, 2021-03-14:
> 已经编译完了，但是在运行时
>  terminate called after throwing an instance of 'std::bad_alloc'
>  what():  std::bad_alloc
>  请问你是否出现过这个问题呀，多谢

没有遇到 😂

zywok, 2021-04-27:
[image]
I'm so sorry for the slow reply,.Compilation errors occur, most of which are caused by PCL library problems.  we have update the PCL library install method and supply the test version link, you can try it .

closed 2021-04-27 by zywok