ISEE-Technology/CamVox #25: 关于内外参问题

HeXu1, 2021-04-21:
您好，非常感谢你们的分享，这个工作非常有意义。
我仔细阅读了你们的文章和代码，我有一点关于内外参的问题想请教。
1 外参标定优化工作：在Calibrating::Run()中优化了best_r_, best_p_, best_y_，优化旋转是为了矫正生成深度图像的偏差，但我好像没发现使用其对激光生成的深度图像进行矫正，是不是我理解的有问题呢？
2 激光生成深度图像的工作：
1）在livox_lidar2depth函数中
这些变量“float fx=1732.78774,fy=1724.88401,cx=798.426021,cy=312.570668; ”是表示相机的内参么，好像与camera.yaml文件中 
Camera.fx: 957.994  
Camera.fy: 955.3280 
Camera.cx: 790.335  
Camera.cy: 250.6631 的参数不太一致，如果不是相机内参的话是什么意思呢。
 2） 这个变量“Eigen::Affine3f sensorPose = (Eigen::Affine3f)Eigen::Translation3f(-0.082984,-0.0221759,0.0727962);”是坐标中心到相机的平移量么，平移对生成深度图影响较小所以在“Calibrating::Run()”中仅优化了旋转么？

期待您的回复，谢谢啦～ 再次感谢你们的分享

zywok, 2021-04-23:
您好，感谢您的建议。这个问题给您造成的困扰，确实是我们在这块没有说明白。

(1)  我们确实准备将优化后的旋转矩阵R保存到.yaml配置文件,然后在生成深度图这块更新使用新的外参矩阵R对深度图进行矫正，所以放射变换矩阵这块只用了原始的Lidar相对于相机的坐标系$ $（3x3)，和一个平移矩阵（3x1).

(2) 这里的fx,fy,cx,cy是我们实时测试使用的相机内参（原因是换过一个相机），而 camera.yaml 里面的是原来采集数据时的相机内参。

(3) 这里的放射矩阵和生成深度图标准使用方式应该是：

   Eigen::Translation3f t； // Lidar->Camera extrinsic translation

   Eigen::Quaternionf q； // Lidar->Camera extrinsic rotation （Expressed by quaternion ）

   **Eigen::Affine3f sensorPose = t*q.toRotationMatrix();**  // Lidar coordinate system affine transformation to the initial pose of the camera coordinate system .

   pcl::RangeImage::CoordinateFrame coordinate_frame = pcl::RangeImage::**CAMERA_FRAME**;     

希望能解决你的困惑，后面我们会把优化后的R更新使用的生成深度这块。再次感谢你的建议！

HeXu1, 2021-04-23:
见到您的回复很高兴，这样算是理顺了，非常感谢。 我应该明白了这个框架的创新点，我简单概括一下：1)用lidar信息代替RGBD深度信息;2)在执行过程中动态对外参的旋转进行微调以达到生成更精准的深度信息。 是这样么～

期待您的更新，我会持续跟进！ 谢谢！

zywok, 2021-04-23:
是这样的，对于之前给你造成的困惑感到很抱歉，我们会继续完善。如有兴趣也可以参考本文作者之一对Lidar和Camera自动标定的的后续优化工作 **https://github.com/hku-mars/livox_camera_calib** ，谢谢。

HeXu1, 2021-04-23:
好的，非常感谢，我的确对标定这一块比较感兴趣 您推荐的论文我看过了 代码还没仔细阅读，我的理解感觉也是一个外参调优的过程。
我用公开数据测试了一下标定效果很好，我自己的数据标定效果不太好。 我感觉这个算法对场景要求有点苛刻，我的理解是：在一个视角横纵向边缘约束丰富并且激光容易提取的场景 目前这个场景对我来说不是很好找～
非常感谢您的建议。

closed 2021-04-27 by zywok