ISEE-Technology/CamVox #28: Dataset problem.

JinMing-Gu, 2021-05-26:
Thank you for your excellent work.
I downloaded the data set CamVox.bag you uploaded, but when I execute the following command:
`rosbag play CamVox.bag`
Display:
`[ INFO] [1622026026.629494531]: Opening CamVox.bag
[FATAL] [1622026026.629722490]: Required 'op' field missing`
Then I execute the following command:
`rosbag info CamVox.bag`
Display:
`ERROR bag unindexed: CamVox.bag. Run rosbag reindex.`
Then I execute the command:
`rosbag reindex CamVox.bag`
After reindexing rosbag, I execute the command again:
`rosbag info CamVox.bag`
Display:
`path:        CamVox.bag
version:     2.0
duration:    6:11s (371s)
start:       May 15 2020 16:14:29.30 (1589530469.30)
end:         May 15 2020 16:20:41.00 (1589530841.00)
size:        15.0 GB
messages:    7436
compression: none [7436/7436 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
topics:      /isee_depth   3718 msgs    : sensor_msgs/Image
             /isee_rgb     3718 msgs    : sensor_msgs/Image`
As shown above, I lost /IMU and /livox_ lidar.
Can you upload the intact rosbag data again?
Thank You!

zywok, 2021-05-29:
Thank you for finding out this issue, please try this link (http://www.zywok.com:20441/index.php/s/ZykmlNVeMLY2rsM) .

garyxu615, 2021-06-01:
Thanks for the excellent work.
I meet the same issue on the bag file. 
I re-download the bag file from the new given link (http://www.zywok.com:20441/index.php/s/ZykmlNVeMLY2rsM) , but this issue still exist. Can you re-check this?
Thank you!

JinMing-Gu, 2021-06-01:
> Thank you for finding out this issue, please try this link (http://www.zywok.com:20441/index.php/s/ZykmlNVeMLY2rsM) .
Thanks for your quick response.
But I found that I couldn't access the link: http://www.zywok.com:20441/index.php/s/ZykmlNVeMLY2rsM
Whether the link is invalid or not?

JinMing-Gu, 2021-06-01:
Thanks for your quick response.
But I found that I couldn't access the link(http://www.zywok.com:20441/index.php/s/ZykmlNVeMLY2rsM) at 16:00 in 20210601
Maybe there is a problem with your server in this time period?

JinMing-Gu, 2021-06-02:
Thanks for your quick response.
I downloaded the dataset from the link(http://www.zywok.com:20441/index.php/s/ZykmlNVeMLY2rsM) again, but the same problem still exists.

zywok, 2021-06-05:
Thank you, I have re-uploaded the Bag package, please try it again.   
OwnCloud link (http://192.168.3.3:20441/index.php/s/bj55H4bgYh2bqSF)
or BaiduNetdisk link (https://pan.baidu.com/s/1O5To1Ic_-EMQ2D4O0kUitA     password：1314 )

mext169, 2021-11-30:
Hello, the link to BaiduNetdisk has expired, can you send it again?

xiaopinghong, 2021-12-01:
> 

sorry for the expired link. we will update shortly.

huiminyuan, 2021-12-02:
We have re-uploaded the Bag package, please try it again. (https://zenodo.org/record/5749452)