ISEE-Technology/CamVox #29: Sensor intrinsic and extrinsic

HengLi-RM, 2021-06-07:
Thank you for sharing the code.
I read the code and found that the sensor extrinsic and intrinsic parameters are unclear.
The camera intrinsic in the code are different from those in yaml. https://github.com/ISEE-Technology/CamVox/issues/25 mentioned that the camera was changed once. So which is the accurate intrinsic?
In addition, it is mentioned in https://github.com/ISEE-Technology/CamVox/issues/25 that the imu information is not obtained from horizon, but from Inertial Sense uINS. So what is the extrinsic from imu to horizon or imu to camera?
So could you provide a yaml file corresponding to the Camvox.bag dataset, including the intrinsic and extrinsic of imu-camera-livox.
Thanks again, sincerely.

JasonWang9805, 2022-11-28:
hello ，did this be solved?

KkCabin, 2024-02-27:
hello ，did this be solved?