ISEE-Technology/CamVox #34: rosrun 运行问题

ningning234, 2021-11-25:
我们看到您团队的CamVox项目的实现效果非常不错，对这一项目非常感兴趣。
感谢您在github上将其开源，我们也试着将其复现。但是我在安装运行过程中，遇到了一些问题无法解决，希望请教一下

我按照步骤安装，但是没装livox-sdk与mvs camera-sdk。我想只运行一下rosbag看看效果。

我打开了roscore ，之后运行 rosrun online camvox Vocabulary/ORBvoc.bin camvox/online/Livox.yaml  时，出现segement failed。

经过调试，我发现错误出现在calibrating.cc文件中，cv::projectPoints(pts_3d, r_vec, t_vec, camera_matrix, distortion_coeff, pts_2d);这个函数出错，
pts_3d, r_vec, t_vec, camera_matrix, distortion_coeff等参数均有对应数值，但是输出的pts_2d值为nan，导致之后的程序无法运行。

由于在rosrun这一步已经出错，我没有运行rosbag。

感谢您的帮助

qiuliuyang-code, 2021-11-25:
我也遇到了一样的问题，我发现是extrinsic_params向量的第四个参数过大，也就是平移过大

chengguangzhi, 2021-12-02:
我也是遇到了同样问题，但我发现问题出现在calibrating.cc文件中``Mat Calibrating::convertByIntensity(const Vector6d &calib_params)``中的``image_project.at<ushort>(point_2d.y, point_2d.x) = intensity``，这一行导致出错。

xieKKKi, 2022-07-21:
我也有这个问题，请问有大佬解决了吗？

xieKKKi, 2022-07-27:
我调试发现是Calibrating.cc里的Calibrating初始化函数里
```
loadParams(strSettingPath_);
```
这句出了问题，奇怪的是，我直接把loadParams函数里的内容复制到Calibrating里，然后把调用loadParams的语句注释，就不会出这个问题了。现在程序应该可以正常运行。
代码如下：
```
Calibrating::Calibrating(string _strSettingPath, string _RGB_path, string _Pcd_path, string _Projection_type, bool _isEnhanceImg, bool _isFillImg) : mbResetRequested(false), mbFinishRequested(false), mbFinished(true), mbStopped(false), mbStopRequested(false), mbNotStop(false), mbCalibrating(false), mbOptimizing(false)
  {
    // initialize parameters
    depth_edge_threshold_ = 80;
    best_r_ = 0;
    best_p_ = 0;
    best_y_ = 0;
    optimize_type = 0;
    loadPcd(_Pcd_path);
    //loadParams(strSettingPath_);

    // loadParams 在上面运行时会出现segement failed，这样写就不会
    {
        cv::FileStorage fSettings(_strSettingPath, cv::FileStorage::READ);
        if (!fSettings.isOpened())
        {
          cerr << "Failed to open settings file at: " << _strSettingPath << endl;
          exit(-1);
        }

        fx_ = fSettings["Camera.fx"];
        fy_ = fSettings["Camera.fy"];
        cx_ = fSettings["Camera.cx"];
        cy_ = fSettings["Camera.cy"];
        k1_ = fSettings["Camera.k1"];
        k2_ = fSettings["Camera.k2"];
        p1_ = fSettings["Camera.P1"];
        p2_ = fSettings["Camera.P2"];
        k3_ = fSettings["Camera.k3"];
        bf = fSettings["Camera.bf"];
        width_ = fSettings["Camera.width"];
        height_ = fSettings["Camera.height"];
        rect_.x = fSettings["Rect.x"];
        rect_.y = fSettings["Rect.y"];
        rect_.width = fSettings["Rect.width"];
        rect_.height = fSettings["Rect.height"];

        mDepthMapFactor_ = fSettings["DepthMapFactor"];
        mDepthMapFactor_ = 1 / mDepthMapFactor_;
    }
```

另外需要注意的是System.cc和Calibrating.cc在这里是直接被编译成动态链接库被调用，如果要修改的话，可以将online文件夹里的CMakeLists.txt文件中的
```
${PROJECT_SOURCE_DIR}/../../lib/libCamvox.so
```
注释掉，并加入：
```
add_library(${PROJECT_NAME} SHARED
        ${PROJECT_SOURCE_DIR}/../../src/System.cc
        ${PROJECT_SOURCE_DIR}/../../src/Tracking.cc
        ${PROJECT_SOURCE_DIR}/../../src/LocalMapping.cc
        ${PROJECT_SOURCE_DIR}/../../src/LoopClosing.cc
        ${PROJECT_SOURCE_DIR}/../../src/ORBextractor.cc
        ${PROJECT_SOURCE_DIR}/../../src/ORBmatcher.cc
        ${PROJECT_SOURCE_DIR}/../../src/FrameDrawer.cc
        ${PROJECT_SOURCE_DIR}/../../src/Converter.cc
        ${PROJECT_SOURCE_DIR}/../../src/MapPoint.cc
        ${PROJECT_SOURCE_DIR}/../../src/KeyFrame.cc
        ${PROJECT_SOURCE_DIR}/../../src/Map.cc
        ${PROJECT_SOURCE_DIR}/../../src/MapDrawer.cc
        ${PROJECT_SOURCE_DIR}/../../src/Optimizer.cc
        ${PROJECT_SOURCE_DIR}/../../src/PnPsolver.cc
        ${PROJECT_SOURCE_DIR}/../../src/Frame.cc
        ${PROJECT_SOURCE_DIR}/../../src/KeyFrameDatabase.cc
        ${PROJECT_SOURCE_DIR}/../../src/Sim3Solver.cc
        ${PROJECT_SOURCE_DIR}/../../src/Viewer.cc
        ${PROJECT_SOURCE_DIR}/../../src/pointcloudmapping.cc
        ${PROJECT_SOURCE_DIR}/../../src/Calibrating.cc
        )
```
最后一行改成
```
target_link_libraries(camvox ${LIBS} ${PROJECT_NAME})
```

wuyuanmm, 2022-08-23:
其实是 Calibrating.cc 中的 strSettingPath_ RGBPath PcdPath_要改成自己工程的路径